DomiStyle/esphome-panasonic-ac #90: Compilig with ESPHOME 2023.4.0 gives some errors

Sashik259, 2023-04-22:
After updating ESPHOME to 2023.4.0 I got some errors while compiling. 
I don't actually understand what it is but I suppose something will stop work in next ESPHOME update

```
In file included from src/esphome/core/controller.h:29,
                 from src/esphome/components/api/api_server.h:4,
                 from src/esphome/components/api/api_connection.h:6,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/climate/climate.h: In constructor 'esphome::climate::Climate::Climate()':
src/esphome/components/climate/climate.h:167:7: warning: 'esphome::climate::Climate::away' is deprecated: away is deprecated, use preset instead [-Wdeprecated-declarations]
  167 | class Climate : public EntityBase {
      |       ^~~~~~~
src/esphome/components/climate/climate.h:193:8: note: declared here
  193 |   bool away{false};
      |        ^~~~
src/esphome/components/climate/climate.h:167:7: warning: 'esphome::climate::Climate::away' is deprecated: away is deprecated, use preset instead [-Wdeprecated-declarations]
  167 | class Climate : public EntityBase {
      |       ^~~~~~~
src/esphome/components/climate/climate.h:193:8: note: declared here
  193 |   bool away{false};
      |        ^~~~
In file included from src/esphome.h:35,
                 from src/main.cpp:3:
src/esphome/components/panasonic_ac/esppac.h: In constructor 'esphome::panasonic_ac::PanasonicAC::PanasonicAC()':
src/esphome/components/panasonic_ac/esppac.h:33:7: note: synthesized method 'esphome::climate::Climate::Climate()' first required here
   33 | class PanasonicAC : public Component, public uart::UARTDevice, public climate::Climate {
      |       ^~~~~~~~~~~
In file included from src/esphome.h:36,
                 from src/main.cpp:3:
src/esphome/components/panasonic_ac/esppac_cnt.h: In constructor 'esphome::panasonic_ac::CNT::PanasonicACCNT::PanasonicACCNT()':
src/esphome/components/panasonic_ac/esppac_cnt.h:19:7: note: synthesized method 'esphome::panasonic_ac::PanasonicAC::PanasonicAC()' first required here
   19 | class PanasonicACCNT : public PanasonicAC {
      |       ^~~~~~~~~~~~~~
src/main.cpp: In function 'void setup()':
src/main.cpp:226:75: note: synthesized method 'esphome::panasonic_ac::CNT::PanasonicACCNT::PanasonicACCNT()' first required here
  226 |   panasonic_ac_cnt_panasonicaccnt = new panasonic_ac::CNT::PanasonicACCNT();
 
```     |

Sashik259, 2023-04-25:
The same in ESPHome 2023.4.1

marcovolt, 2023-04-27:
Same error

sygys, 2023-04-30:
@DomiStyle Hey Dominik, could you report back to this thread when this is resolved. (if ever) so that i know that i can safely update.

Thanks man. Im still very grateful of the work you did. My ac's have been running for a few years now with your solution and are rock solid!

Herbi-1966, 2023-05-05:
Same here, somtimes, 8 reboots after flashing - in safe mode, it works for a while.
@DomiStyle: thank you for your effort, I really appreciate your work :-)

closed 2023-05-19 by Sashik259