DomiStyle/esphome-panasonic-ac #91: Temperature Sensor Decimals

filou01, 2023-05-08:
Hi!
Its really a great integration and i love it. 

My problem is that the indoor unit sends over UART the temperature only in 0.5 degree steps, so i decided to add an external sensor. This also works, but again i only see 0.5 degree steps in the frontend and in my graphs. 

[image]

The VERY_VERBOSE log output shows that is seems as the ESP sends 0.1 degree values to HA. Is there any other value in the API Integration that tells HA that the values are only in 0.5 degree steps available during initialization?

[11:02:51][V][panasonic_ac.cz_tacg1:248]: Polling AC
[11:02:51][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[11:02:52][V][panasonic_ac:247]: RX: 70.20.00.2B.80.A0.39.00.00.00.00.00.00.2B.00.00.00.00.80.80.FF.17.0C.FF.80.80.FF.80.22.00.01.00.00.00.FE (35)
[11:02:52][D][climate:392]: 'Klima' - Sending state:
[11:02:52][D][climate:395]:   Mode: OFF
[11:02:52][D][climate:403]:   Custom Fan Mode: Automatic
[11:02:52][D][climate:409]:   Custom Preset: Normal
[11:02:52][D][climate:412]:   Swing Mode: OFF
[11:02:52][D][climate:415]:   Current Temperature: 23.11°C
[11:02:52][D][climate:421]:   Target Temperature: 21.50°C
[11:02:52][V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[11:02:52][V][json:056]: Size after shrink 216 bytes
[11:02:52][VV][api.service:242]: send_climate_state_response: ClimateStateResponse {
  key: 87801859
  mode: CLIMATE_MODE_OFF
  current_temperature: 23.1067
  target_temperature: 21.5
  target_temperature_low: 0
  target_temperature_high: 0
  legacy_away: NO
  action: CLIMATE_ACTION_OFF
  fan_mode: CLIMATE_FAN_ON
  swing_mode: CLIMATE_SWING_OFF
  custom_fan_mode: 'Automatic'
  preset: CLIMATE_PRESET_NONE
  custom_preset: 'Normal'
}
[11:02:52][V][component:204]: Component panasonic_ac.climate took a long time for an operation (0.09 s).
[11:02:52][V][component:205]: Components should block for at most 20-30ms.

Thank you and regards,
Christoph

DomiStyle, 2023-05-08:
> Is there any other value in the API Integration that tells HA that the values are only in 0.5 degree steps available during initialization?

There's `temperature_step`, `target_temp_step` and `current_temp_step` in the ESPHome/Home Assistant API.

I only set `target_temp_step` so it shouldn't affect the current temperature in any way unless the default in ESPHome is `0.5`, so you might try setting the step to `0.1` anyway?

For more details see here: https://esphome.io/components/climate/index.html#base-climate-configuration

All of these are only for visual display though, so you might want to check the raw state in your Home Assistant dev tools.

filou01, 2023-05-08:
The Dev Tools shows also a value rounded to 0.5I guess it‘s a behavior of the Climate ESPHome integration in HA?Von meinem iPhone gesendet

filou01, 2023-05-08:
The Dev Tools shows also a value rounded to 0.5I guess it‘s a behavior of the Climate ESPHome integration in HA?

[image]

which platform should i use here:
climate:
  - platform: ...
    visual:
      min_temperature: 18
      max_temperature: 25
      temperature_step: 0.1

filou01, 2023-05-08:
Got it!

climate:
  - platform: panasonic_ac
    type: cnt
    visual:
      temperature_step:
        current_temperature: 0.1
        target_temperature: 0.5

… did the trick. Thank you for your hint!

closed 2023-05-08 by filou01