DomiStyle/esphome-panasonic-ac #92: How to debug?

JshGrn, 2023-06-19:
Hi, I am fairly new to this I have connected up the unit to my CN-CNT terminal and connected to HA however the temp says 0c and if I set it to cool or another mode it swaps back to off.

The command line says 
[image]

What can I do to debug the issue? Is there a way I can check if the ESP is actually communicating correctly?

JshGrn, 2023-06-19:
I am just continuously getting:

[17:04:21][V][panasonic_ac.cz_tacg1:248]: Polling AC
[17:04:21][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)

I am using the board 'az-delivery-devkit-v4'

DomiStyle, 2023-06-19:
Do you get any RX lines? If not check if your wiring is correct, try swapping the RX/TX pins. The ESP is not getting any data back.

JshGrn, 2023-06-19:
No, the board I am using labels RX and TX as GPO1 and 3, so I even tried adjusting in the config. 

The pin out is here, I have also tried swapping them round.

Additionally, I have tried 16/17 (which are not labeled as RX/TX).

I am not using a logic converter as this board should have it in-built (not getting hot either)

DomiStyle, 2023-06-19:
Try 16/17 or 9/10 (both in config and physically). The RX/TX pins on the ESP32 are used for the logger.

[image]

JshGrn, 2023-06-19:
Where do you get that from? I am not seeing UART on those pins? I am very new to ESP so maybe I am getting something wrong here.

I just retested 16/17 and flipped but still nothing 

https://cdn.shopify.com/s/files/1/1509/1638/files/ESP32_DevKit_C_V4_Pinout.pdf?v=1615364529

DomiStyle, 2023-06-19:
That's the ESPHome documentation:

https://esphome.io/components/logger.html
https://esphome.io/components/uart.html

Unfortunately it's hard to debug these issues remotely so you will have to do some measurements on your own to check if your wiring is correct.

JshGrn, 2023-06-19:
Just to clarify something here, my unit has WiFi module built in, and I was following the guide (https://github.com/DomiStyle/esphome-panasonic-ac/blob/master/README.CZTACG1.md), maybe this is causing an issue? I saw others have both working.

DomiStyle, 2023-06-19:
If your connector looks identical to the CN-CNT connector on that page you're on the correct guide.

It's possible to connect both of them at the same time, yes.

JshGrn, 2023-06-19:
It works now... very odd, maybe something wasn't connected up right.

My next issue is making the test wire I have into a connector so will now go away and find the connectors I need to connect it 'securely' :)

I am going to be running 4 of these in the same house, will they clash with the same local alias (panasonic-ac.local) or should I set them up manually to be their own host name.

DomiStyle, 2023-06-19:
The hostname does not matter, it will simply be harder to identify them when doing network scans. You can always flash them with the same hostname manually and give them a hostname later on via OTA.

closed 2023-06-19 by JshGrn