DomiStyle/esphome-panasonic-ac #94: Not working on CS-Z35XKEW / probalby new protocol/change?

jonasf21, 2023-06-21:
I've got a panasonic CS-Z35XKEW (Etherea). It appears its no longer possible to replace the existing DNSK-P11 because the protocol seems to have changed. The module itself now also locks a bit different (different labels/style but still reads DNSK-P11).
I've captured the TX/RX and this is what it does:

[20:36:29][V][panasonic_ac:245]: TX: 5A.09.10.01.00.05.01.30.01.00.01.54 (12)
[20:36:29][V][panasonic_ac:247]: RX: 5A.09.10.81.00.D6.00.01.30.01.00.01.00.01.00.33.00.80.E2.01.00.85.42.04.00.86.62.2E.00.88.62.01.00.A0.E2.01.00.A1.E2.01.00.A4.E2.01.00.A5.E2.01.00.B0.E2.01.00.B2.E2.01.00.BB.42.01.00.BE.42.01.02.20.62.01.02.21.62.01.02.31.E2.01.02.32.62.01.02.33.E2.01.02.34.E2.01.02.35.E2.01.02.36.E2.01.02.42.82.01.02.7D.E2.01.02.7E.42.01.02.7F.42.02.02.80.42.02.02.81.42.02 (120)
[20:36:29][D][panasonic_ac.dnskp11:280]: Dropping invalid packet (checksum)
[20:36:29][V][panasonic_ac:247]: RX: 02.82.42.02.02.83.42.02.02.84.42.02.02.89.C2.0C.02.8A.C2.06.02.8B.C2.06.02.8C.C2.0E.02.91.42.02.02.92.C2.02.02.96.42.02.02.97.42.02.02.98.42.02.02.99.42.02.02.9A.42.02.02.9B.42.02.02.A0.C2.01.02.A1.C2.07.02.A2.C2.02.02.A3.C2.04.02.A4.C2.02.02.A5.C2.02.02.A8.42.02.02.A9.C2.03.02.AA.E2.02.02.FF.42.40.FC (101)
[20:36:29][W][panasonic_ac.dnskp11:249]: Dropping invalid packet (header)
[20:36:29][D][panasonic_ac.dnskp11:721]: Resending previous packet
[20:36:29][V][panasonic_ac:245]: TX: 5A.09.10.01.00.05.01.30.01.00.01.54 (12)
[20:36:30][V][panasonic_ac:247]: RX: 5A.09.10.81.00.D6.00.01.30.01.00.01.00.01.00.33.00.80.E2.01.00.85.42.04.00.86.62.2E.00.88.62.01.00.A0.E2.01.00.A1.E2.01.00.A4.E2.01.00.A5.E2.01.00.B0.E2.01.00.B2.E2.01.00.BB.42.01.00.BE.42.01.02.20.62.01.02.21.62.01.02.31.E2.01.02.32.62.01.02.33.E2.01.02.34.E2.01.02.35.E2.01.02.36.E2.01.02.42.82.01.02.7D.E2.01.02.7E.42.01.02.7F.42.02.02.80.42.02.02.81.42.02 (120)
[20:36:30][D][panasonic_ac.dnskp11:280]: Dropping invalid packet (checksum)
[20:36:30][E][component:113]: Component panasonic_ac.climate was marked as failed.

The RX packets are switching between those with invalid header and invalid checksum. The RX packets do not change (besides those with invalid header vs. those with invalid checksum). So its not noise or interference.

Any ideas?

DomiStyle, 2023-06-21:
The handshake is just a 1:1 replay of what my DNSK-P11 sent when I captured it years ago.
I never bothered actually finding out what the fields do.

As a quick test you could try disabling the checksum test to see if the handshake runs through: https://github.com/DomiStyle/esphome-panasonic-ac/blob/master/components/panasonic_ac/esppac_wlan.cpp#L32

It pretty much does not matter what the AC sends at this point, it just matters that the AC initializes successfully so that it starts accepting commands.

If that does not work, do you have a CNT port as well? You can connect it there as well.

jonasf21, 2023-06-21:
Just commected Line 32/33 of esppac_wlan.cpp and recompiled. Now i am getting this:

[22:25:00][V][panasonic_ac:247]: RX: 66.08.00.01.01.F6 (6)
[22:25:00][W][panasonic_ac.dnskp11:613]: Received unknown packet during initialization
[22:25:01][D][panasonic_ac.dnskp11:204]: Starting handshake [1/16]
[22:25:01][V][panasonic_ac:245]: TX: 5A.00.00.06.00.00.A0 (7)
[22:25:01][V][panasonic_ac:245]: TX: 5A.01.00.09.00.00.9C (7)
[22:25:01][V][panasonic_ac:247]: RX: 66.08.00.01.01.F6 (6)
[22:25:01][W][panasonic_ac.dnskp11:613]: Received unknown packet during initialization
[22:25:01][V][panasonic_ac:247]: RX: 66.08.00.01.01.F6 (6)
[22:25:01][W][panasonic_ac.dnskp11:613]: Received unknown packet during initialization
[22:25:02][V][panasonic_ac:247]: RX: 66.08.00.01.01.F6 (6)

jonasf21, 2023-06-21:
I've tested some settings and it seems most work (swing mode etc). However and most important temperature change does not work:
[22:40:37][D][climate:010]: 'Panasonic AC' - Setting
[22:40:37][D][climate:039]:   Target Temperature: 18.50
[22:40:37][V][panasonic_ac.dnskp11:104]: Requested temperature change
[22:40:37][V][panasonic_ac:245]: TX: 5A.2B.10.08.00.09.01.01.30.01.01.00.31.01.25.CF (16)
[22:40:37][V][panasonic_ac:247]: RX: 5A.2B.10.88.00.09.03.01.30.01.01.00.31.01.25.4D (16)
[22:40:37][V][panasonic_ac.dnskp11:456]: Received command ack

It always reverts back to the previous setting - i cant change the temperature.

DomiStyle, 2023-06-21:
Is that with or without checksum enabled?

Try setting the temperature via IR remote and take note of the differences in the packages you receive. The temperature should be in there in hexadecimal, just divide it by 2 and you should get the correct temperature.

All info I collected about the protocol is available in this folder so you can compare: https://github.com/DomiStyle/esphome-panasonic-ac/tree/master/protocol

jonasf21, 2023-06-21:
Its without checksum. I changed the temp by IR remote to 21.5:
[22:47:03][V][panasonic_ac:247]: RX: 5A.0D.10.0A.00.19.00.01.30.01.05.00.A1.01.44.00.B0.01.41.02.20.01.43.02.31.01.2B.02.33.01.43.19 (32)

So it should be 21.5x2 = 43...in hex 2B, which is in the string. But the field after it is 02, in your doc it says 00 (but n/a, so not sure if that matters).

DomiStyle, 2023-06-21:
Hmm, looks similar but much longer than a usual response to a temperature change.

Depending on how far you want to go you will probably have to get a packet capture with a logic analyzer from the original wifi module or switch to the CNT port once it arrives. Hopefully nothing changed there.

> so maybe something related to nanoex changed.

Definitely, they rebranded and moved this feature around like 5 times already.

jonasf21, 2023-06-21:
Sorry, i guess it also changed some other settings as i turned the AC on via HA.
I again changed temp (from previous 21.5 set via infrared) to 21 now:
[23:09:40][V][panasonic_ac:247]: RX: 5A.0E.10.0A.00.09.00.01.30.01.01.02.31.01.2A.E4 (16)

DomiStyle, 2023-06-21:
Can you try again if setting the temperature works? That report looks identical to mine.

jonasf21, 2023-06-21:
Still doesnt work:

[23:19:55][D][climate:039]:   Target Temperature: 19.00
[23:19:55][V][panasonic_ac.dnskp11:104]: Requested temperature change
[23:19:55][V][panasonic_ac:245]: TX: 5A.81.10.08.00.09.01.01.30.01.01.00.31.01.26.78 (16)
[23:19:56][V][panasonic_ac:247]: RX: 5A.81.10.88.00.09.03.01.30.01.01.00.31.01.26.F6 (16)
[23:19:56][V][panasonic_ac.dnskp11:456]: Received command ack

What seems to be different is that the string sent via the IR has a different "30" block. The IR one is
30.01.01.02 while the one from esphome is 30.01.01.00 (so with 00 at the end). From your controller protocol doc it appears you expect this to be always 00...so this might be the cause?

DomiStyle, 2023-06-21:
You could try poking around in the code, the function that receives commands from Home Assistant is here: https://github.com/DomiStyle/esphome-panasonic-ac/blob/master/components/panasonic_ac/esppac_wlan.cpp#L62

The temperature gets set here: https://github.com/DomiStyle/esphome-panasonic-ac/blob/master/components/panasonic_ac/esppac_wlan.cpp#L105

The `set_value` function fills a queue made from a `key` and `value`. `0x31` is the key for target temperature.

This is the function where the final packet is assembled and sent: https://github.com/DomiStyle/esphome-panasonic-ac/blob/master/components/panasonic_ac/esppac_wlan.cpp#L621

It just loops through the queue and adds all parameters that were set like this:

`KEY 0x01 VALUE 0x00`

If you just want to replace the 00 at the end with 02 (whatever that means, I have no idea) you can change this line: https://github.com/DomiStyle/esphome-panasonic-ac/blob/master/components/panasonic_ac/esppac_wlan.cpp#L640

The function is fairly well documented, at least with what I know.

Also, it should be noted that the syntax between the IR commands and the wifi module commands differs (there are 2 files in the protocol documentation), the key and values are identical but the structure of the packet is different and mostly unknown.

jonasf21, 2023-06-21:
Thanks, i'll try and see if that helps.

Since the code does not work with checksumming enabled - how is it calculated? I seem to be unable to find info about that.

DomiStyle, 2023-06-21:
The package is verified in this function: https://github.com/DomiStyle/esphome-panasonic-ac/blob/master/components/panasonic_ac/esppac_wlan.cpp#L231

The checksum is calculated here: https://github.com/DomiStyle/esphome-panasonic-ac/blob/master/components/panasonic_ac/esppac_wlan.cpp#L273

Essentially it just sums together every byte (including the checksum), if the sum is 0 the checksum is valid.

In the command from above: `5A.81.10.08.00.09.01.01.30.01.01.00.31.01.26.78`
If you sum all of them together you get `0x0200`, since we're working with a single byte the `0x02` overflows and you are left with `0x00`, which means the packet above is valid.

Might also be an idea to create a branch that disable checksumming during initialization and enables it after it bruteforced its way through the initialization, that way you can keep checksumming enabled for regular packets.

jonasf21, 2023-06-21:
Ahh thanks for the insight on the checksum, got it now.

As from the first post it appears the checksum is totally of. 
5A.09.10.81.00.D6.00.01.30.01.00.01.00.01.00.33.00.80.E2.01.00.85.42.04.00.86.62.2E.00.88.62.01.00.A0.E2.01.00.A1.E2.01.00.A4.E2.01.00.A5.E2.01.00.B0.E2.01.00.B2.E2.01.00.BB.42.01.00.BE.42.01.02.20.62.01.02.21.62.01.02.31.E2.01.02.32.62.01.02.33.E2.01.02.34.E2.01.02.35.E2.01.02.36.E2.01.02.42.82.01.02.7D.E2.01.02.7E.42.01.02.7F.42.02.02.80.42.02.02.81.42.02 
actually results in 0x1DF9. No wonder it gets dropped and it doesnt pass init. They must have changed something there.

I'll poke around a bit and see if i can get the temp change working somehow (now that i know its probably not the checksum).

jonasf21, 2023-06-21:
Yay, that was quick. I changed 00 to 02 from the "30" block and guess what...i can set temperature now :)

Maybe you can check if the 02 works on your AC too....and it could be changed in the code. Otherwise it would require testing
or maybe add a special setting to the yaml to for this?

jonasf21, 2023-06-22:
Update:
Actually the answer to the checksum problem is in my first post - yay. The packet is not missing the checksum, its just split in half.
The header states the packet is D6 long, so 214 byte - however the code currently only accepts 128 byte messages (BUFFER_SIZE). I changed the buffer size (esppac.h) to 240 byte and increased the READ_TIMEOUT to 240. I then re-enabled checksumming and its working now! (i tested with read_timeout at 40 & 100 but it didnt work, so i went near the max - maybe there is a better spot).

JshGrn, 2023-06-30:
I have these and the CNT port works flawlessly alongside the original WiFi module (so you can keep the app running too).

jonasf21, 2023-06-30:
@jshgrn i do not want to use the CNT port - its old and outdated (newer units always use CN-WLAN). I also wanted to replace the DNSPK11 module completly (dont want any foreign/external cloud). So i replaced the DNSPK11 and used its connection (CN-WLAN) - and for this there are changes required.

JshGrn, 2023-06-30:
I know, but why does that concern you re it being outdated? There is no functionality that it doesnt provide that the CN-WLAN does...

jonasf21, 2023-06-30:
I assume the CN-CNT will be removed from future units since its no longer needed (reason: cost). Also as for feature wise comparision: i am not sure this is true. The etherea device have alot more protocol info than previous gen devices and its not fully decoded - so cant be sure if CN-CNT really would pass all that info. Anyway - the main reason people are doing the change is to remove the panasonic cloud and for that you have to open the unit and remove the DNSPK11 module - while there you can just use the connector there anyway (and dont need to go for CN-CNT).

JshGrn, 2023-06-30:
Removing CN-CNT... why is this a concern, are you updating your units every month? I am not sure I agree with your thoughts about them removing it, its there for a reason, its very likely they will keep it.

What features do you think are missing? I currently use this with the Etherea range (3 x same model 1 x 2.5kw) and they all work perfectly. All features from the app work with this except position of vents, not sure that the WIFI version has this either as I think its missing from the package.

I see your logic re removing the wifi module, although I don't understand why you would over using the CN-CNT currently, I'd just turn it off. Its still nice to have the app *sometimes*.

jonasf21, 2023-06-30:
No i dont touch the AC any month but i'd like to keep this project/code up2date, so future users will be able to use it (on either connector).
Afaik the power consumption calculation is different between CN-CNT and CN-WLAN. However its unsupported on CN-WLAN as of now (i might start working on that later). 
I dont trust any device i dont have full control of - specially if its connected to some type of cloud. As that the DNSPK11 is unusable for me and has to go. 
Lets not try to start a discussion here regarding what connector is better or if panasonic's cloud is trustworthy etc. Everyone can decide that themselves. I just want the code fixed, so people can choose what they prefer - i am adding features not cutting them.

JshGrn, 2023-06-30:
I respect that!

For me - I use the CNT, the power consumption isn't something I have paid much attention to.

The cloud solution they offer is terrible and I do suffer from constant disconnects throughout the year, maybe I one day will go to the WLAN port if there is reason for me to do so, right now there is not.

edalberg, 2023-11-01:
Very good solution that works perfectly on the older pumps. 

I have 3 new pumps and is testing now on a Panasonic CZ25WKE. On this I get error messages such as:

[07:24:47][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[07:24:47][V][json:058]: Size after shrink 232 bytes

@jonasf21 and @DomiStyle:  Could this have something to do with "buffersize" and "read_timeout" in esppac.h?

Here is my entire log:

[07:24:41][W][component:214]: Component panasonic_ac.climate took a long time for an operation (0.10 s).
[07:24:41][W][component:215]: Components should block for at most 20-30ms.
[07:24:42][VV][api.connection:120]: Sending keepalive PING...
[07:24:42][VV][api.service:037]: send_ping_request: PingRequest {}
[07:24:42][VV][api.service:549]: on_ping_response: PingResponse {}
[07:24:46][V][panasonic_ac.cz_tacg1:252]: Polling AC
[07:24:46][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[07:24:46][V][panasonic_ac:247]: RX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[07:24:46][W][panasonic_ac.cz_tacg1:362]: Received unknown fan speed
[07:24:46][W][panasonic_ac.cz_tacg1:476]: Received unknown mild dry value
[07:24:46][D][climate:380]: 'Warmepumpe-Leiligheta' - Sending state:
[07:24:46][D][climate:383]: Mode: OFF
[07:24:47][D][climate:391]: Custom Fan Mode: Unknown
[07:24:47][D][climate:397]: Custom Preset: Normal
[07:24:47][D][climate:400]: Swing Mode: OFF
[07:24:47][D][climate:403]: Current Temperature: -83.00°C
[07:24:47][D][climate:409]: Target Temperature: 0.00°C
[07:24:47][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[07:24:47][V][json:058]: Size after shrink 232 bytes
[07:24:47][VV][api.service:242]: send_climate_state_response: ClimateStateResponse {
  key: 1694556500
  mode: CLIMATE_MODE_OFF
  current_temperature: -83
  target_temperature: 0
  target_temperature_low: 0
  target_temperature_high: 0
  unused_legacy_away: NO
  action: CLIMATE_ACTION_OFF
  fan_mode: CLIMATE_FAN_ON
  swing_mode: CLIMATE_SWING_OFF
  custom_fan_mode: 'Unknown'
  preset: CLIMATE_PRESET_NONE
  custom_preset: 'Normal'
}
PS uses the CN-CNT interface

Any idea?

JshGrn, 2023-11-01:
@edalberg I have this working on the same models (1 2.5kw and 2 3.5kw) as you using the CN-CNT interface without any issues, I haven't seen that error.

edalberg, 2023-11-01:
@JshGrn Strange! I have 2 more at the cabin, so I will test them out at the weekend.

edalberg, 2023-11-20:
> Very good solution that works perfectly on the older pumps.
>
>
>
> I have 3 new pumps and is testing now on a Panasonic CZ25WKE. On this I get error messages such as:
>
>
>
> [07:24:47][V][json:038]: [...]

The problem was a bad soldering!