DomiStyle/esphome-panasonic-ac #96: Unable to get it working on a CS-Z25UD3EAW (ducted unit)

CamielFolmer, 2023-07-10:
Hi, I've been trying for a while now to get a CS-Z25UD3EAW working with this project. 

Besides not being able to power the esp from the CN-CNT port (I meaure 5v on the pins, but the esp won't boot, but it does with external power to the esp), i have the following in the logging:

18:37:59 | [V] | [panasonic_ac.cz_tacg1:252] | Polling AC
18:37:59 | [V] | [panasonic_ac:245] | TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
18:37:59 | [V] | [panasonic_ac:247] | RX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
18:37:59 | [W] | [panasonic_ac.cz_tacg1:362] | Received unknown fan speed
18:37:59 | [W] | [panasonic_ac.cz_tacg1:476] | Received unknown mild dry value
18:37:59 | [D] | [climate:378] | 'Panasonic AC' - Sending state:
18:37:59 | [D] | [climate:381] | Mode: OFF
18:37:59 | [D] | [climate:389] | Custom Fan Mode: Unknown
18:37:59 | [D] | [climate:395] | Custom Preset: Normal
18:37:59 | [D] | [climate:398] | Swing Mode: OFF
18:37:59 | [D] | [climate:401] | Current Temperature: -83.00°C
18:37:59 | [D] | [climate:407] | Target Temperature: 0.00°C
18:37:59 | [V] | [json:036] | Attempting to allocate 512 bytes for JSON serialization
18:37:59 | [V] | [json:056] | Size after shrink 224 bytes
18:37:59 | [V] | [component:204] | Component panasonic_ac.climate took a long time for an operation (0.08 s).
18:37:59 | [V] | [component:205] | Components should block for at most 20-30ms.
18:38:04 | [V] | [panasonic_ac.cz_tacg1:252] | Polling AC
18:38:04 | [V] | [panasonic_ac:245] | TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
18:38:04 | [V] | [panasonic_ac:247] | RX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)

Does this mean there is communication with the unit?

the ESP is connected to the CN-CNT port (replaced a cz-tag1).

Is there anything I can do to troubleshoot this further, any suggestions on what might be wrong?

DomiStyle, 2023-07-10:
Did you accidentally connect RX and TX together? The ESP is receiving exactly what it is sending, which is strange.

CamielFolmer, 2023-07-11:
Physically only dupont connections are being used (no soldering), and no logic level convertor. So no possibility I've connected tx and rx.. Picture:
[image]

CamielFolmer, 2023-07-11:
Like this?
[image]

SirAceAir, 2023-09-04:
Did you solve this Problem?
I have exactly the Same Situation with the Same LOG

CamielFolmer, 2023-09-14:
No stil the same issue, I would really like to get this working but not sure where to start with these errors

arcticShadow, 2023-10-08:
Based on the earlier feedback that the esp was receiving the same data it was sending out it sounds like there may be a short in some wiring somewhere. If you have access to a multi meter, check if you have continuity between your rx and tx wires  on your connector. (There shouldn't be) (do this with the connectors disconnected from the AC) 

Also check if the receiving pins that correspond to the Rx and Tx on the AC unit have continuity (they shouldn't) 

Disclaimer: I Havn't used this software, nor do I have this particular AC unit. This just sounds like an interesting problem that may be fixable

CamielFolmer, 2025-08-25:
I've never been able to solve it, still would like to solve it though. I've given up, don't know where to start.