DomiStyle/esphome-panasonic-ac #97: external temperature sensor

sygys, 2023-07-12:
Hey Dominik / @DomiStyle ,

Its been some time since we spoke. My ac's run like a charm for quite some time now with your code. I remember scrolling through your files and excel sheet back in the day when i started using this to see which bits and bites of the long code do stuff. I also remember that in those files there were some bits of code which did nothing or atleast you couldnt figure out what they were for. 

Running the ac for some time now i notice that the measured indoor temperature of the AC unit is pretty off from the real temperature. I was wondering if maybe some of these bits of code are there to communicate with the ac for hooking up an external temp sensor.

Im not even sure if this is an option for these AC's because i havent found any online. But it would be awsome to let HA post the temperature to the AC from an external sensor. Though im not sure if you are still working on this project and are willing to do some digging into this. I dont believe there are bits of code the AC is communicating that dont mean anything. Every part of the code the ac's is posting back is for something i guess...

Also one more thing. I noticed that since updating the AC's esp to the latest esphome version the logs dont show the communication of the AC and the long bitcode anymore. Do you know a way to get this communication back? 

Just wanted to say that this piece of code is by far the best work i have ever seen someone do in home assistant. This integration of the AC locally is GOLD!

DomiStyle, 2023-07-12:
> Running the ac for some time now i notice that the measured indoor temperature of the AC unit is pretty off from the real temperature. I was wondering if maybe some of these bits of code are there to communicate with the ac for hooking up an external temp sensor.

I doubt there's such a feature in there since Panasonic would most likely advertise it.

> Also one more thing. I noticed that since updating the AC's esp to the latest esphome version the logs dont show the communication of the AC and the long bitcode anymore. Do you know a way to get this communication back?

It still does when verbose logging is enabled.

> Just wanted to say that this piece of code is by far the best work i have ever seen someone do in home assistant. This integration of the AC locally is GOLD!

Good to hear it has been of use to people besides me.

froqstar, 2026-01-08:
While researching this topic I stumbled upon the "Intesis" smart home adapters by HMS networks such as [the PA-AC-KNX-1i](https://www.hms-networks.com/p/inknxpan001i000-panasonic-etherea-ac-units-to-knx-interface-with-binary-inputs?tab=tab-support) which is even referred to [by Panasonic themselves](https://www.aircon.panasonic.eu/EE_et/control/connectivity-1/) where they claim:

> Use the air conditioner ambient temperature or the one measured by a KNX temperature sensor or Thermostat.

Now that got me interested how they do that, if it's not officially supported through Panasonic first party products? 
Turns out they tell us how, on page 36 of the [user manual](https://hmsnetworks.blob.core.windows.net/nlw/docs/default-source/products/intesis/manuals-and-guides---manuals/user-manual-inknxpan001i000.pdf?sfvrsn=f03250d7_13):

> the following formula applies for calculation of real `Control_ Setpoint_Temperature` sent to the AC unit:  
>
>> “AC Setp. Temp.” = “KNX Setp. Temp.” - (“KNX Amb. Temp.” - “KNX Setp. Temp.”)/2
>
>> AC Setp. Temp.: AC indoor unit setpoint temperature
>> KNX Amb. Temp.: Ambient temperature provided from KNX
>> KNX Setp. Temp.: Setpoint temperature provided from KNX
>
> As an example, consider the following situation:
> User wants: 19ºC (“KNX Setp. Temp.”)
> User sensor (a KNX sensor) reads: 21ºC (“KNX Amb Temp.”)
> In this example, the final setpoint temperature that INKNXPAN001I000 will send out to the indoor unit (shown in “AC Setp. Temp.”) will become 19ºC – (21ºC - 19ºC)/2 = 18ºC. This is the setpoint that will actually be requested to Panasonic unit.

If this solution is good enough to be used in a 500€ product endorsed by Panasonic and hopefully got quite some field testing by now, it may be worth a shot? 

User @War3333 [seems to own this device](https://github.com/DomiStyle/esphome-panasonic-ac/issues/86#issuecomment-1567239393) so may be able to confirm?

sygys, 2026-01-08:
I have been using something like this in home assistant for a few years now. Just make a PID controller that sends the temperature to the ac based the room temp. The ac will keep our living room heated or cooled with a precision of around 0.1 degree. I have been wondering though how bad this is for the Ac itself setting a new temperature every minute or so