DomiStyle/esphome-panasonic-ac #98: Compatibility with

skellycode, 2023-07-18:
Hello!

I've been looking at adding some smarts to my existing in-roof ducted panasonic AC system and came across this library.
I'm wondering whether or not it would be compatible with my unit?

My in-roof unit is the `S-71PE1R5A` which is presently controlled by the `CZ-RTC4`.  All of the panasonic marketing materials do not mention the ability to add one of their wifi modules as the unit is from ~2016 and seems to pre-date the wifi products.  I've been up in the roof and taken some pictures of the PCB to see if I could identify whether the right connectors exist but I'm not certain - could someone more knowledgeable than I please have a look?

I can see that there is a `T10` connector, but according to the manual, this seems to only be used for switching... does that mean its not the same as the T10 that is required for this library to work?  It also seems to be a 6-pin connector, when from what I've seen, split systems have a 5-pin?
See page 23 (brochure page 45) https://www.panasonic.com/content/dam/Panasonic/support_manual/Air_Conditioner/English/FY2015_PAC_BROCHURE.pdf

[image]

In the service manual the only reference to `T10` I can see is:

[image]

I'm not able to find any references to `CN-CNT`

Thank you in advance!

p.s. here is an overview of the whole board:

[image]

DomiStyle, 2023-07-18:
If there's no `CN-CNT` or `CN-WLAN` connector on your indoor unit there is unfortunately no way to get this library to work.

skellycode, 2023-07-18:
Thanks @DomiStyle.  Another google brings up no references to `CN-CNT` on this board, so I guess I'm out of luck.

Thank you, and best of luck with the project 😄

closed 2023-07-18 by skellycode