DomiStyle/esphome-panasonic-ac #102: A little over my head - question about prebuilt esp boards.

jack3308, 2023-10-29:
Hi there! I've been  peaking at this on and off for a few months now. Just moved into a new place that has a bunch of the cz-tag wifi adapters on units all throughout the place. I don't really have the ability (yet) or equipment to put these together from components, but I have a (very little) bit of experience using things like the m5stack atom series of esp dev boards for things like this. I was wondering if one of those would work for this project. I know I'm probably asking a silly question, but the terminology and language in the instructions is just a step or two above my head and I'm having a hard time finding resources that help me understand better.

codyc1515, 2024-01-09:
I would also purchase one if it would work out of the box.

HA-TB303, 2025-03-14:
This should help. The DIY route is really not that hard: [https://www.ingeniousmakers.com/p11](https://www.ingeniousmakers.com/p11)