DomiStyle/esphome-panasonic-ac #104: MQTT data

jemarr, 2023-12-05:
Hi,

Thanks for your great job, it's awesome.

I've installed it on a Wemos D1 mini board controlling my CS-Z60UD3E and it's running well. I can control AC thought ESPHome web page.

My domotic system is based on Node-Red, so to control your software parameters I need MQTT or HTTP commands, so are there any way to do that without Home Assistant ?

Thanks in advance.

tomhri, 2023-12-06:
#  api:

# Example configuration entry
mqtt:
  broker: 192.168.0.150
  #  username:
  #  password:

tomhri, 2023-12-06:
Had the same issue, look at 
https://esphome.io/components/mqtt.html

jemarr, 2023-12-06:
Thanks for your answer.

Now I got MQTT client running.

Now I just need the topics to publish data to AC and topics to subscribe data from AC.

Do you know which are them ?

I just know those that appears on debug window:

[17:47:17][C][mqtt.select:028]: MQTT Select 'Panasonic AC Vertical Swing Mode':
[17:47:17][C][mqtt.select:029]:   State Topic: 'AC/select/panasonic_ac_vertical_swing_mode/state'
[17:47:17][C][mqtt.sensor:028]: MQTT Sensor 'Panasonic AC Outside Temperature':
[17:47:17][C][mqtt.sensor:032]:   State Topic: 'AC/sensor/panasonic_ac_outside_temperature/state'

Bot not the Mode or target temperature, etc

Thanks in advance.

closed 2024-01-05 by DomiStyle