DomiStyle/esphome-panasonic-ac #109: Read vertical swing mode via REST api?

pontusfischer, 2024-01-18:
Hi, I activated the esphome web server by adding this line to my yaml:
web_server:
  port: 80

I can now request the status via (http://panasonic-ac.local/climate/panasonic_ac
However the vertical swing state (swing/auto/down/down-center/center/up center/up) is missing. 
How can I add this? In web_server.cpp I can (apparently?) only add supported traits to the json. Any way around this?

DomiStyle, 2024-01-19:
I don't really use the web server so I can't tell you for certain but I don't set anything to determine what gets exposed to the web interface.

If it's missing it might be an esphome limitation.

pontusfischer, 2024-01-19:
Hi and thanks for your reply. I may have figured out another way. I will post the solution here when finished. For now I have the following problem, in the yaml file I have another component that gives me e.g. the new vertical swing mode to set. 
Lets say I have the following codeline in the yaml:

str::string newVertSwingMode = data.substr(15).c_str();
I cannot do this:
id(my_climate).set_vertical_swing_select = newVertSwingMode;

my_climate is the same Panasonic ac component as in your example yaml, just with an added ID.
in esppach.h I find this:
set_vertical_swing_select(select::Select *vertical_swing_select)
So this function obviously doesn't accept an std::string like e.g. "down", but wants an argument of type select::Select. 
What is the correct way to set e.g. the vertical swing mode of the AC in the yaml file?
I ask for vertical swing mode in particular since the official ESPhome climate component doesn't know about this feature. Setting the target temperature is easy. Vertical swing, not so much...

DomiStyle, 2024-01-19:
The vertical swing is a select element in ESPHome (select::Select). The `set_vertical_swing_select` only sets a reference to that select, it does not set the value.

The value itself is set here: https://github.com/DomiStyle/esphome-panasonic-ac/blob/a55ed6616d56437e2f3a367fb4719e2ff77ad4f6/components/panasonic_ac/esppac.cpp#L103

pontusfischer, 2024-01-19:
I see, thanks!

pontusfischer, 2024-01-19:
void PanasonicAC::update_swing_vertical(const std::string &swing) { 
Is actually declared protected, but I guess changing this to public won't brake anything :D
I'll post my solution when finished, this might be of interest for people who are not using HomeAssistant and want to interface to another smart home system without using MQTT.

pontusfischer, 2024-01-21:
I can now read all parameters and set most of them. Setting (writing) the vertical swing mode is still giving me some trouble though, could you give me any pointers on how to do this with a lambda function in the YAML?

I can set the vertical swing mode to auto like this:
auto call = id(my_climate).make_call();
call.set_swing_mode(CLIMATE_SWING_VERTICAL);

However the different vertical swing modes (up/center/down etc) are not configurable using the set_swing_mode function. So what is the proper way to do this?