DomiStyle/esphome-panasonic-ac #111: [INFO] Some ESP32 feedback

jellewie, 2024-02-05:
Not a bug report or an issue, just wanted to give some feedback to help the README and such alone.
I do have some minor questions/bugs but I will post those separately so they can be tracked
Thanks for making this, it works well and straight out of the box for me!

------------
I have a model with Both CN-CNT and CN-WLAN ports. The DNSK-P11 module was already installed in the CN-WLAN port.
[image]
Since I use the CN-CNT port I needed to configure "type: cnt".

I do not have 12V at the CNT connector. (it's floating)

Since an ESP32 is completely 5V tolerant, it can just work without a level shifter (The 3.3V data from the ESP is higher to the 5V than 0V on the Airco receiver side, so that also doesn't even mind it)
I just literally connected it like this to the ESP32:
[image]
I uploaded the example code, and it just worked.

TheDK, 2024-05-11:
That was almost too easy - ESP32 NodeMCU DevKit C with a CN105 to Jumper Pigtail. Just connected, flashed and works instantly.
No Level Shifter - I’ll report back should the 5V on the UART bus kill the ESP…

[image]

jellewie, 2024-05-11:
I have been running this ever since on both my units. **No problem at all so far** with the hardware.

Software I found a specific bug with my unit, but that's nothing to do with the electrical and stements above

jellewie, 2024-05-11:
.

Lauriz7, 2024-06-17:
> Not a bug report or an issue, just wanted to give some feedback to help the README and such alone. I do have some minor questions/bugs but I will post those separately so they can be tracked Thanks for making this, it works well and straight out of the box for me!
> 
> I have a model with Both CN-CNT and CN-WLAN ports. The DNSK-P11 module was already installed in the CN-WLAN port. ![image](https://private-user-images.githubusercontent.com/1545223/302344755-13f7d658-7132-461a-928b-83536568e24b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg2NTQ3OTQsIm5iZiI6MTcxODY1NDQ5NCwicGF0aCI6Ii8xNTQ1MjIzLzMwMjM0NDc1NS0xM2Y3ZDY1OC03MTMyLTQ2MWEtOTI4Yi04MzUzNjU2OGUyNGIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MTdUMjAwMTM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Yzc5ZTU0ZWEzMTQ0NTAyYjRhNzNlNzVmNjM1MTg4N2MwNWIzOTdhNTdjN2U5NTUzMWUzY2FjNzhkMzM5YzdhOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.EPHsu8XkH4PMWJddjMo_Y3uLmQqaq772tSkrw3QS_DA) Since I use the CN-CNT port I needed to configure "type: cnt".
> 
> I do not have 12V at the CNT connector. (it's floating)
> 
> Since an ESP32 is completely 5V tolerant, it can just work without a level shifter (The 3.3V data from the ESP is higher to the 5V than 0V on the Airco receiver side, so that also doesn't even mind it) I just literally connected it like this to the ESP32: ![image](https://private-user-images.githubusercontent.com/1545223/302348052-f110c8d7-369c-4d3a-acb0-e13022a9fae1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg2NTQ3OTQsIm5iZiI6MTcxODY1NDQ5NCwicGF0aCI6Ii8xNTQ1MjIzLzMwMjM0ODA1Mi1mMTEwYzhkNy0zNjljLTRkM2EtYWNiMC1lMTMwMjJhOWZhZTEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MTdUMjAwMTM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmM3NzFkNDcxNTU5NmRjMGU0OGI1MDUzMGQyNTkzMjlmYmQ1ZDYwZjc2MjFlMTA0NGE2OGJhZjI2Y2Q3M2YyYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.9fBr1OpHE9AG1NokcX1auqkHSKSeeYtPZ6iAWcnizvQ) I uploaded the example code, and it just worked.
> 
> The Airco now uses 0.25W more with the ESP32. The Airco + WiFi was 0.65W, and now with the ESP32 included its about 0.9W. It is of course possible to remove the WiFi module to have a lower Idle load again. ![image](https://private-user-images.githubusercontent.com/1545223/302353018-d8d8bbd8-e606-4a97-89b2-09e498f2bea4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg2NTQ3OTQsIm5iZiI6MTcxODY1NDQ5NCwicGF0aCI6Ii8xNTQ1MjIzLzMwMjM1MzAxOC1kOGQ4YmJkOC1lNjA2LTRhOTctODliMi0wOWU0OThmMmJlYTQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MTdUMjAwMTM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjJiMmUyOGQ0ZTAzYTczYzZlNjFjMzIwMDJjMDY0OGNhMWY5OTliZjVmYjEzMGM5NzI2MjNkZmUyNmFjMzRhMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.yvcIEMSkT44lejpNLRKdWuY_8W4MlWuRjDkY03azLCA)

> Not a bug report or an issue, just wanted to give some feedback to help the README and such alone. I do have some minor questions/bugs but I will post those separately so they can be tracked Thanks for making this, it works well and straight out of the box for me!
> 
> I have a model with Both CN-CNT and CN-WLAN ports. The DNSK-P11 module was already installed in the CN-WLAN port. ![image](https://private-user-images.githubusercontent.com/1545223/302344755-13f7d658-7132-461a-928b-83536568e24b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg2NTQ3OTQsIm5iZiI6MTcxODY1NDQ5NCwicGF0aCI6Ii8xNTQ1MjIzLzMwMjM0NDc1NS0xM2Y3ZDY1OC03MTMyLTQ2MWEtOTI4Yi04MzUzNjU2OGUyNGIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MTdUMjAwMTM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Yzc5ZTU0ZWEzMTQ0NTAyYjRhNzNlNzVmNjM1MTg4N2MwNWIzOTdhNTdjN2U5NTUzMWUzY2FjNzhkMzM5YzdhOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.EPHsu8XkH4PMWJddjMo_Y3uLmQqaq772tSkrw3QS_DA) Since I use the CN-CNT port I needed to configure "type: cnt".
> 
> I do not have 12V at the CNT connector. (it's floating)
> 
> Since an ESP32 is completely 5V tolerant, it can just work without a level shifter (The 3.3V data from the ESP is higher to the 5V than 0V on the Airco receiver side, so that also doesn't even mind it) I just literally connected it like this to the ESP32: ![image](https://private-user-images.githubusercontent.com/1545223/302348052-f110c8d7-369c-4d3a-acb0-e13022a9fae1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg2NTQ3OTQsIm5iZiI6MTcxODY1NDQ5NCwicGF0aCI6Ii8xNTQ1MjIzLzMwMjM0ODA1Mi1mMTEwYzhkNy0zNjljLTRkM2EtYWNiMC1lMTMwMjJhOWZhZTEucG5nP1gtQW [... cut ...] ODliMi0wOWU0OThmMmJlYTQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MTdUMjAwMTM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjJiMmUyOGQ0ZTAzYTczYzZlNjFjMzIwMDJjMDY0OGNhMWY5OTliZjVmYjEzMGM5NzI2MjNkZmUyNmFjMzRhMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.yvcIEMSkT44lejpNLRKdWuY_8W4MlWuRjDkY03azLCA)

does it still work for you?

jellewie, 2024-06-17:
@Lauriz7 yes this is still running flawlessly for me

danielcherubini, 2024-06-22:
Thanks to this, I just got mine up and running. Works flawlessly.

Wimll, 2024-11-07:
Thanks for starting this thread and sharing this info! I check the documentation and wondered why a ESP32 with 5v in directly was not possible. This thread confirmed this, which is great! Maybe we could add a PR to add this additional info to the readme? 😄

jellewie, 2024-11-07:
@Wimll Yhea we could. That's why I shared the info here in the first place for.

I have 2 airco's with this mod and they have been working perfectly.

My airco's have an esp size space at the outside of the 230V area. I just soldered 4 wires to the PCB at the back of the connector. And pull them though the home. I guess the space I now use is ment for a second wifi unit or some sorts? I don't have a picture on hand

Wimll, 2024-11-07:
> I have 2 airco's with this mod and they have been working perfectly.

Thanks! I will have a look inside my units, still have to do the install. If mine also has this open space, I think this mod is meant to be done anyway? 😋

jellewie, 2024-11-07:
[image]
Found a picture of my instalment

jeroenvdd, 2024-11-07:
this mights now sound completly stupid  but:

do I see this correctly and is this just a basic esp32 without any other hardware and thus not in need of the boards / c3 mini like og project?

jellewie, 2024-11-07:
@jeroenvdd yes you are correct. The ESP32 is 5V tolerant. So just need to connect the 4 wife's shown in my drawing in the order I drew them in. Works best to power the AC down if you do so

jeroenvdd, 2024-11-07:
> @jeroenvdd yes you are correct. The ESP32 is 5V tolerant. So just need to connect the 4 wife's shown in my drawing in the order I drew them in. Works best to power the AC down if you do so

dang I only have 1 wife :D 

other than that good to hear I have some testing to do this evening

rickardfransson, 2024-11-09:
So, A ESP32 and some cables and then this repo downloaded to the ESP32?.
Thats all? Its then connectable till HA?

willie013, 2024-11-11:
> So, A ESP32 and some cables and then this repo downloaded to the ESP32?.
> Thats all? Its then connectable till HA?

Yes, that's exactly what this whole discussion is about 😜

kevindegrootgrave, 2024-11-14:
I just ordered 4 ESP32's and the precrimped cables mentioned before by @willie013. It's working absolutely perfect, thanks for all the info. I've just got one problem: my AC only supports vertical swing, but since vertical AND horizontal swing are mentioned in the components/panasonic/ac folder it is still selectable in the AC card. Is there a way to completely turn the horizontal swing option off?
Also, I'm using Node Red to control my AC's. With the Comfort Cloud HACS integration I can control swing mode by sending {"swing_mode":"Up"} to the climate.set_swing_mode action. With this ESP Home solution that doesn't work, because available swing modes are off, both, horizontal and vertical. Can anybody tell me what the correct command is I should use now?

jellewie, 2025-01-21:
@kevindegrootgrave I believe there is an open ESPhome issue for that swing option. I have the same, but if you try and select it the AC unit will correct your mistake and the dashboard will update back to its (new) current state.

riegojerey, 2025-03-10:
Good thing I was read this. I will try now! Could you provide your yaml code also?

jellewie, 2025-05-08:
@asmigala, it's important to understand that the Wemos S2 Mini is not 5V tolerant. Without using a level shifter -as I previously suggested- connecting 5V signals directly can cause damage and lead to unreliable behavior. This is very likely the root cause of the issues you're experiencing.

The ESP32 doit devkit v1 (what i suggested) is 5V tolerant, that's why I suggested tried and suggested the board.

iznmastermind, 2025-05-23:
The CN-CNT and CN-WLAN works the same? Can i have both esp and CZTAG running simultaneously? CN-WLAN uses 5v also?

jellewie, 2025-05-23:
@iznmastermind i don't quite understand your question.

But if you have an airco with both connections. You are able to leave the WiFi module in the CN-WLAN port, and add an esp32-doit board to the CN-CNT port.

Due to the WiFi unit now being useless to me and for power reasons. I switched to only having the ESP32 connected, I have runned both at the same time before.

iznmastermind, 2025-05-23:
Ok got it. Thanks a lot

TheDK, 2025-05-23:
Just FYI: My first ever ESP32 (out of ~20-30 across my house) died ~1 day after the AC was turned on again after being without power for ~6 months. I measure 0 impedance between 3.3v and GND, meaning the voltage regulator on the board apparently died. Replaced the ESP32 and works fine again - will monitor and report back if it happens again. Another AC did not show the same problem.

jellewie, 2025-06-23:
@duczz That's an really weird suggestion/question.

Just simply use an ESP32-DOIT-DEVKIT board (which is 5V tolerant).
Or use any ESP that is 3.3V with an level shifter.

I just made this topic to give people the feedback that an DOIT board can work without any other hardware requirements. I have 2 of these going and they are still rock solid