DomiStyle/esphome-panasonic-ac #113: Current Temperature showing as nan

kenchy, 2024-02-22:
Hey, I'm on the long road away from the home assistant comfort cloud integration. I've got it working as a direct replacement for a CZ-TACG1, the only issue I've found so far is the current temperature isn't coming back correctly for the unit:

[20:39:50][V][panasonic_ac.cz_tacg1:252]: Polling AC
[20:39:50][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[20:39:50][V][panasonic_ac:247]: RX: 00.20.40.28.80.A0.10.04.00.40.00.00.00.28.00.00.00.00.0D.80.FF.0D.80.FF.80.80.FF.80.01.00.00.C0.00.00.14 (35)
[20:39:50][V][panasonic_ac.cz_tacg1:186]: Outside temperature is not supported
[20:39:50][D][climate:380]: 'froompac' - Sending state:
[20:39:50][D][climate:383]:   Mode: OFF
[20:39:50][D][climate:391]:   Custom Fan Mode: Automatic
[20:39:50][D][climate:397]:   Custom Preset: Quiet
[20:39:50][D][climate:400]:   Swing Mode: OFF
**[20:39:50][D][climate:403]:   Current Temperature: nan°C**
[20:39:50][D][climate:409]:   Target Temperature: 20.00°C
[20:39:50][W][component:214]: Component panasonic_ac.climate took a long time for an operation (0.06 s).
[20:39:50][W][component:215]: Components should block for at most 20-30ms.

Any ideas?

DomiStyle, 2024-02-22:
Can you post your yaml file? Make sure you do not have `current_temperature_sensor` set there.

I can see the temperature just fine in the packet sent by your AC.

kenchy, 2024-02-23:
OK so that was caused by me totally not understanding what that sensor was for. I had a current temperature set, I'm not sure why. Presumably that is so you can pull a sensor from home assistant.

Thanks for responding so quickly all working fine now.

Can I ask why the outside temperature isn't supported? It's there in the cursed comfort cloud.

DomiStyle, 2024-02-23:
> Can I ask why the outside temperature isn't supported? It's there in the cursed comfort cloud.

Your outside temperature is set to 0x80 (decimal 128) which has been used by Panasonic to say there is no outside temperature sensor on the unit so far.

If you can confirm that your outdoor unit has a sensor and if you can find your outside temperature by comparing the packets with verbose logging during night/day I might be able to add support for it.

kenchy, 2024-02-23:
Well apparently there is an outside temperature sensor and it is reporting the temperature to HA:

[image]

So all good!

Again thank you for the help.

closed 2024-02-27 by kenchy