DomiStyle/esphome-panasonic-ac #115: Help to uncomment remote function

lollo78, 2024-03-06:
Hi, thanks for this magic project!

My Panasonic AC (NKE series), CZ-TAC interface, use the [A75C3887 remote.](https://www.electrohogar.net/62-thickbox_default/cwa75c3887-a75c3887-a75c4179.jpg)

What features I have to uncomment in the esphome config? 

Ok for **mild_dry_switch** and **econavi_switch** because they are supported, but **eco_switch** needs to be uncomment? 
Last doubt is about **nanoex_switch**: NanoEX (this is the name of the technology in my AC) is the same things of NanoE-G?
Thanks a lot

lollo78, 2024-04-24:
Hi, I tried all the extra remote functions, but no one works. Switches turn off immediately and nothing happens on the AC.
For example, Econavi is supported on my AC but enabling it in esphome It didn't works.
Any advice?
Thx

DomiStyle, 2024-04-24:
You can enable verbose logging and toggle econavi back and forth via the remote. You should see something changing in the polling response the ESP gets.

The currently known fields are documented here: https://github.com/DomiStyle/esphome-panasonic-ac/tree/master/protocol/cztacg1

lollo78, 2024-04-24:
With Verbose and toggle Econavi on and off with the Panasonic remote I see this:

```
C:\>esphome run climatizzatore-stanzetta.yaml
INFO ESPHome 2024.4.1
INFO Reading configuration climatizzatore-stanzetta.yaml...
INFO Updating https://github.com/DomiStyle/esphome-panasonic-ac.git@None
INFO Generating C++ source...
INFO Compiling app...
Processing climatizzatore-stanzetta (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32@5.4.0)
------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (5.4.0) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
Building in release mode
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\api\api_connection.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\api\api_frame_helper.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\api\api_pb2.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\api\api_pb2_service.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\api\api_server.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\api\list_entities.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\api\proto.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\api\subscribe_state.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\api\user_services.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\captive_portal\captive_portal.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\climate\climate.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\climate\climate_mode.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\climate\climate_traits.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\dallas\dallas_component.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\dallas\esp_one_wire.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\esp32\core.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\esp32\gpio.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\esp32\preferences.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\json\json_util.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\logger\logger.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\logger\logger_esp32.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\logger\logger_esp8266.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\logger\logger_host.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\logger\logger_libretiny.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\logger\logger_rp2040.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\md5\md5.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\mdns\mdns_component.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\mdns\mdns_esp32.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\mdns\mdns_esp8266.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\mdns\mdns_host.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\mdns\mdns_libretiny.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\mdns\mdns_rp2040.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\network\util.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\ota\ota_backend_arduino_esp32.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\ota\ota_backend_arduino_esp8266.cpp.o
Compiling .pioenvs\climatizzatore-stanzetta\src\esphome\components\ota\ota_backend_arduino_libretiny.cpp.o
[... 481 lines cut ...]
[21:01:05][V][panasonic_ac:247]: RX: 70.20.44.2C.80.A0.1D.00.00.00.00.00.48.2E.00.00.00.00.80.80.FF.80.80.FF.80.80.FF.80.00.00.00.00.00.00.D0 (35)
[21:01:05][V][panasonic_ac.cz_tacg1:176]: Current temperature is not supported
[21:01:05][D][climate:396]: 'Climatizzatore Stanzetta' - Sending state:
[21:01:05][D][climate:399]:   Mode: HEAT
[21:01:05][D][climate:407]:   Custom Fan Mode: Automatic
[21:01:05][D][climate:413]:   Custom Preset: Normal
[21:01:05][D][climate:416]:   Swing Mode: HORIZONTAL
[... cut ...]
[21:01:10][D][climate:396]: 'Climatizzatore Stanzetta' - Sending state:
[21:01:10][D][climate:399]:   Mode: HEAT
[21:01:10][D][climate:407]:   Custom Fan Mode: Automatic
[21:01:10][D][climate:413]:   Custom Preset: Normal
[21:01:10][D][climate:416]:   Swing Mode: HORIZONTAL
[21:01:10][D][climate:419]:   Current Temperature: nan°C
[21:01:10][D][climate:425]:   Target Temperature: 22.00°C
[21:01:10][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[21:01:10][V][json:058]: Size after shrink 240 bytes
```