DomiStyle/esphome-panasonic-ac #119: New CS-Z25ZKEW, no reaction on RX/TX

wunderbaum, 2024-06-27:
Hello,
since a few days, I have a new CS-Z25ZKEW. This is a model with an empty plug and the wlan somewhere inside.
So, I thought, I can connect like this: [(https://github.com/DomiStyle/esphome-panasonic-ac/issues/111)]
I have an ESP32-C3-Super-Mini in use. At the 12V I have measured 14V. But this I do not need.
Type I choosed is cnt, when I choose wlan, I get timeouts in the console.
In my model, I cannot even see the included wlan-module.
[image]
I switched RX/TX, no success. 
What can I do to dive deeper into a possible solution?

This is my config:
`substitutions:
  name: esphome-web-cb0e7c
  friendly_name: Klima Schlafzimmer

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/firmware/esphome-web/esp32c3.yaml@v2
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

uart:
  tx_pin: 20
  rx_pin: 21
  baud_rate: 9600
  parity: EVEN

external_components:
  source: github://DomiStyle/esphome-panasonic-ac
  components: [panasonic_ac]

climate:
  - platform: panasonic_ac
    # For CZ-TACG1
    type: cnt

    # For DNSK-P11
    # type: wlan

    name: Panasonic AC
    horizontal_swing_select:
      name: AC Horizontal Swing Mode
    vertical_swing_select:
      name: AC Vertical Swing Mode
    outside_temperature:
      name: AC Outside Temperature
    nanoex_switch:
      name: Panasonic AC NanoeX Switch
`

profigriller, 2024-06-29:
This is my wiring and my yaml. Please ignore the relay and correspondig wires at the bottom. I had issues with the uart and tried different ports. Maybe this helps.

[image]
`substitutions:
  devicename: "fgfgfgf"
  long_devicename: "ESPthings Panasonic Climate Interface"
  pcb_version: "ESP32SAZDelivery"
  
esphome:
  name: "${devicename}"
  name_add_mac_suffix: false ## if 'false' use different names/hostnames for each device!
  comment: "${long_devicename} ${pcb_version}"
  platform: ESP32
  board: nodemcu-32s
  project:
    name: "esphome.pana-climate"
    version: "${pcb_version}"

external_components:
  - source: github://DomiStyle/esphome-panasonic-ac
    components: [panasonic_ac]

wifi:
  ssid: YYYYY
  password: XXXXXX
  ap:
    ssid: "${devicename} Hotspot"
    password: ZZZZZZ

    
captive_portal:
    
web_server:
  port: 80
  
logger:
  level: INFO

ota:
## uncomment the line below if you wish to use an ota-password
# password: !secret esphome_ota_password

uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  id: ac_uart
  baud_rate: 9600
  parity: EVEN
  
## revisions below 220722-01 use GPIO12
status_led:
  pin: GPIO13
 
climate:
  - platform: panasonic_ac
    id: redBox
    type: cnt   # For CZ-TACG1 use type: cnt / For DNSK-P11 use type: wlan
    name: "${long_devicename}"
    vertical_swing_select:
      name: "${long_devicename} Vertical Swing Mode"
    horizontal_swing_select:
      name: "${long_devicename} Horizontal Swing Mode"
    outside_temperature:
      name: "${long_devicename} Outside Temperature"
    nanoex_switch:
      name: Panasonic AC NanoeX Switch
    current_power_consumption:
      name: Panasonic AC Power Consumption

`

wunderbaum, 2024-07-01:
> had issues with the uart and tried different ports.

Many thanks!
That was the easy solution! I changed the UART-GPIOs and now it works. 

`uart:
   tx_pin: GPIO8
   rx_pin: GPIO9
   id: ac_uart
   baud_rate: 9600
   parity: EVEN`

tomba0815, 2024-07-07:
@wunderbaum
Can you tell me the exact name of the connector for the cnt socket on the Pana, or can you send me a link to the Aliexpress article?
I bought "JST PH 2.0 5PIN" plugs, but they don't fit into the socket.
Could you also post your working config?
Have you deactivated the built-in WLAN module or is it running in parallel without any problems?

At the moment I can't get a connection to my CS-Z25ZKEW, I think it's the plug that doesn't have the right contact.

Many thanks
[image]

wunderbaum, 2024-07-07:
Hello @tomba0815 
I have ordered these:
https://a.aliexpress.com/_Eui8PR5

Take care to choose 5P version. 
It works in parallel to the internal module.

tomba0815, 2024-07-09:
Hello @wunderbaum ,
thank you very much for your support :)
I have now ordered the plug and also tested the connection with a DIY solution. It also works so far.

Thomas

Harnas, 2025-04-18:
In ESP32-C3, UART can be used on any pins due to IO Mux https://docs.espressif.com/projects/arduino-esp32/en/latest/tutorials/io_mux.html . Problem was only on older ESP chips, where hardware UART was on was used to firmware upload.

I have successfully ran this project with my CS-Z35ZKEW with following UART config:
`
uart:
   tx_pin: GPIO3
   rx_pin: GPIO4
   id: ac_uart
   baud_rate: 9600
   parity: EVEN
`
This config is handy in ESP32-C3 Super Mini, because those pins are next to power pins. So only one pin header row is required to solder.