DomiStyle/esphome-panasonic-ac #128: [Feature Request] Icons for the fan speeds and setups

ParalaX002, 2024-11-16:
Hi,

First, thanks a lot for this software! I've been switching my units to this, after yet another issue with the Panasonic cloud. But no more thanks to you!

So, just one cosmetic request, if it is even possible. It would be great to have icons for all the possibles actions. I mean, here is a snapshot of the Panasonic comfort cloud integration in home assistant:
[image]
Same with this code:
[image]
Also, setup has no icon. Here is a proposition:
[image]

How could we achieve this?

Thanks in advance!

Hoedi1030, 2024-12-10:
@ParalaX002 How did you manage to get the options for fan speed (1...5 / Auto) and the preset selection (none / quiet / powerful)?

Like, what lines of code do i need to add to the ac.yaml to have these options when i visit the IP-address of my ESP.

Thank you for any help.

d96moe, 2025-01-21:
> Ok, I went ahead and tried something: https://github.com/ParalaX002/esphome-panasonic-ac
> 
> I use the home assistant default definition, however I reduced the number of fan speed to 3 (low, medium and high), because home assistant doesn't have enough granularity...

Nice. I was looking for the same, as I would need the pre-defined fan modes in Home Assistant. Looking at your changes, it seems you only changed the CNT interface, not the WLAN one, right?

d96moe, 2025-01-22:
> [@ParalaX002](https://github.com/ParalaX002) How did you manage to get the options for fan speed (1...5 / Auto) and the preset selection (none / quiet / powerful)?
>
> Like, what lines of code do i [...]

If I understand the implementation correctly, the fan speed and presets are only published to Home Assistant through the esphome climate implementation and are not controllable default from the esp web interface. It may be possible to expose them in the web interface in some way. Here is someone with the same question: https://community.home-assistant.io/t/how-do-you-update-climate-preset-fan-mode-etc-without-ha/546402

ParalaX002, 2025-01-22:
Hi, sorry for the delay :( And nice for the fork :)

I remember I tried to play with the fan diffuse and stuff, but it was not rendering nicely if I recall correctly. I think it is because there isn't any difference in icon, so in the end, that's not working well.

The real problem is that to get those icon, we need to use the default home assistant climate widget, which would gain to have more option. Or have ESPHome do something about it.

The AC Panasonic that use internet don't have this limitation because they can create their own climate entity :s

d96moe, 2025-01-23:
No worries 😀
I see what you mean. The modes are showing up in what I think is in climate::CLIMATE_FAN_MODES enum order that bring complete comfusion..

[image]

ParalaX002, 2025-01-24:
Ah yes that was the issue. Anyway, let's be honest, I think 3 fan mode are more than enough, and on my units anyway, fan mode don't really work anyway so...

Hoedi1030, 2025-02-16:
> > [@ParalaX002](https://github.com/ParalaX002) How did you manage to get the options for fan speed (1...5 / Auto) and the preset selection (none / quiet / powerful)?
> > Like, what lines of code do i need to add to the ac.yaml to have these options when i visit the IP-address of my ESP.
> > Thank you for any help.
> 
> [@Hoedi1030](https://github.com/Hoedi1030) If I understand the implementation correctly, the fan speed and presets are only published to Home Assistant through the esphome climate implementation and are not controllable default from the esp web interface. It may be possible to expose them in the web interface in some way. Here is someone with the same question: https://community.home-assistant.io/t/how-do-you-update-climate-preset-fan-mode-etc-without-ha/546402

@d96moe Thanks for the reply, sadly i could not manage to expose the fan modes in the web interface. As I did not plan to be stuck with only controlling my ac via the rather crude looking esp web interface, I hookep up the device via MQTT to my home automation system (i am using [iobroker](https://www.iobroker.net)) and to my surprise, the fan modes and everything else was accessible and controllable (see screenshot below). So that was a first success.

[image]

I then continued by building a custom GUI for my ac, which turned out quite nicely imo (i "stole" the icons from the panasonic comfort cloud app =) ).

[image]

So it's possible to make use of all the functions (including fan speed and preset selection) even without having a home-assistant installation.

Hope this may be helpful for anyone else having a similar situation as i have.