DomiStyle/esphome-panasonic-ac #131: Can't send or receive anything with m5stack atom s3 lite

bdog720, 2024-12-13:
I had this working on a generic esp32 but I bought and m5stack atom s3 lite for it's small form factor so I could fit it in the AC unit. I've got the m5stack to power on but I can't send or receive anything. I'm attempting to use the 4 pin which states g2 and g1. I've tried gpio1 and gpio2 but nothing. I can't seem to find any reliable information on which gpio ports these use. Can anyone help?

I have more details here on the home assistant forum https://community.home-assistant.io/t/esphome-m5stack-atom-s3-lite-controlling-air-conditioner/809609

bdog720, 2025-01-17:
> Hey [@bdog720](https://github.com/bdog720), have you had any success? I’m trying to do the same and would love to hear if you’ve gotten any further.

No unfortunately I didn't get anywhere. I'm going to be trying again but with https://esphome.io/components/climate/haier.html and see if I get any further

nguyenquangduc, 2025-02-05:
Hello, have you connected your esp 32 s3 to control the air conditioner? I'm also facing the same situation and don't have a solution yet. Thank you

bdog720, 2025-02-05:
I couldn't find a solution so I ended up just using a generic esp32 board

zupo, 2025-04-20:
I tried with my Atom S3 Lite today, and it works 🎉 

I used the Groove connector and set the following:

```
uart:
  tx_pin: GPIO1
  rx_pin: GPIO2
  id: ac_uart
  baud_rate: 9600
  parity: EVEN
```

I have the Panasonic TZ20ZKE model.

nguyenquangduc, 2025-04-21:
i drew a pcb myself and am using gpio 17 for tx and 18 for rx on esp32 s3 but still no response have you tried using these 2 gpios

zupo, 2025-04-21:
No, I tried GPIO1 and GPIO2 first, and it worked, so didn't wanna spend more time on it.

bdog720, 2025-05-11:
Ok I finally got this working on the groove connector, thanks @zupo for the posting that it worked for you.  To confirm here are the esp32 and uart settings that ended up working for me, I swear I tried this before and it don't work but a win is a win
```
esp32:
  variant: ESP32S3
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
```

```
uart:
  tx_pin: GPIO1
  rx_pin: GPIO2
  baud_rate: 9600
  parity: EVEN
```

zupo, 2025-05-11:
Great to hear it worked for you in the end! Could probably close the issue now as well?

closed 2025-05-11 by bdog720