DomiStyle/esphome-panasonic-ac #135: Esp32-s3

nguyenquangduc, 2025-02-05:
Has anyone in the group used the esp32-s3 to control panasonic air conditioners? I configured it but got no response.

Cougar, 2025-05-10:
I'm using S2 which is basically the same but two less performant. Are you sure your wiring is correct?

nguyenquangduc, 2025-05-12:
yes . i am sure my wiring is ok i used pin 17 18 on s3 to communicate but it failed but the same wiring on esp32s is successful

Cougar, 2025-05-12:
I guess that You are not talking about chip pins but board pin numbers? Is it ESP32 S3 DevKitC 1 or anything else? What was the successful board type? I use ESP32s2 mini and just random GPIO ports (16 and 18) that were easiest to wire with level converter. Try some other ports that are not wired with default UART TX and RX.