DomiStyle/esphome-panasonic-ac #136: Switch off ambient temperature sensor

m3tc0n, 2025-02-13:
The ambient temperature sensor in the indoor unit of our Panasonic air conditioner indicates an excessive ambient temperature. Think of this as 4 degrees too high. This causes se air conditioner to switch off too soon while the room is not yet warm enough.

Does anyone know if it is possible to turn off the temperature sensor? Possibly via ze integration?

m3tc0n, 2025-02-16:
Great work! Can I already use it in the yaml file?

m3tc0n, 2025-02-16:
So it would look like this?

```
substitutions:

devicename: "panasonic-airco-01"

long_devicename: "Airco-01"

pcb_version: "230902"

 

esphome:

name: "${devicename}"

name_add_mac_suffix: false ## if 'false' use different names/hostnames for each device!

comment: "${long_devicename} ${pcb_version}"

 

esp32:

board: esp32-c3-devkitm-1

framework:

type: arduino

 

external_components:
  source: github://CSchlipp/esphome-panasonic-ac@temperature-offsets
  components: [panasonic_ac]
  refresh: 0d

 

web_server:

port: 80

 

# Enable logging

logger:

 

# Enable Home Assistant API

api:

encryption:

key: "redacted"

 

ota:

- platform: esphome

password: "redacted"

 

wifi:

output_power: 8.5

ssid: "redacted"

password: "WachtwoordWifiNetwerk"

 

# Enable fallback hotspot (captive portal) in case wifi connection fails

ap:

ssid: "Panasonic-Airco Fallback Hotspot"

password: "Tijdelijkewifi01"

 

captive_portal:

 

uart:

tx_pin: GPIO7

rx_pin: GPIO6

id: ac_uart

baud_rate: 9600

parity: EVEN

## It possible to wire an optional temperature sensor

## DS18B20 1-wire Temperature Sensor

#dallas:

# - pin: GPIO01

#sensor:

# - platform: dallas

# # Check the dallas sensor address in the log file

# # For more info: https://esphome.io/components/sensor/dallas.html

# address: 0x540000001524be28

# name: "$long_devicename Temperature"

# id: temp

switch:

- platform: restart

name: "$long_devicename Restart"

 

climate:

- platform: panasonic_ac

type: cnt

### or change "cnt" into "wlan" for DNSK-P11 ###

name: "${long_devicename}"

vertical_swing_select:

name: "${long_devicename} Vertical Swing Mode"

outside_temperature:

name: "${long_devicename} Outside Temperature"

current_power_consumption:

name: Panasonic AC Power Consumption

#set the offset according to your needs, allowed values: -5 to 5 
    current_temperature_offset: 0
    outside_temperature_offset: 0

## If the ac does not report a current temperature (CZ-TACG1 only) uncomment

## the line below and point to the appropriate sensor

#

# current_temperature_sensor: temp

#

## Since Panasonic ACs support different features you can comment out the lines below if applicable:

#

# eco_switch:

# name: Panasonic AC Eco Switch

# nanoex_switch:

# name: Panasonic AC NanoeX Switch

# mild_dry_switch:

# name: Panasonic AC Mild Dry Switch

# econavi_switch:

# name: Econavi switch

# current_power_consumption:

# name: Panasonic AC Power Consumption

```

m3tc0n, 2025-02-16:
OK, same is for the external components. It is a copy paste thing.

m3tc0n, 2025-02-16:
OK, I have edited my yaml file and installed it on the board. 

I have set current_temperature_offset to -5.

The ambient temperature sensor is now showing 17 degrees celsius in stead of 22. That looks promising!

m3tc0n, 2025-02-16:
You are welcome!

I think a offset of -5 degrees celsius will fit my needs. 

I will test it tomorrow.

willie013, 2025-02-24:
Great work @CSchlipp ! I will add that a offset might works, but the temprature sensor is just unreliable. Ideally the temperature sensor used to control the heating of the AC can be set to an external temp sensor, such that it is much more stable than the internal temperature sensor