DomiStyle/esphome-panasonic-ac #140: [Info] Ready to use controllers for sale in EU

senhus, 2025-04-30:
@DomiStyle Please delete this issue this is not ok with you 👍 

This project sparked the passion in me to get started with creating my own electronics as the cloud version was driving me crazy. Even with a electronics and IT background it was frustrating to get started. 
It was the same experience with compact presence sensors not being available in EU. so I decided to establish a hobby  business selling this stuff that I learn to make with the goal of:

1. Making custom sensors and controllers that otherwise not available for purchase and is a outcome of open-source community only
2. Create a "Build Your Own" section which I share my designs such as 3d designs for the sensors and link to open source resource and eventually video tutorial
3. Give back a portion of the profit to open source contributors in Home Assistant.

So if creating your own controller from scratch is no go for you, you can purchase it here:
https://www.senhus.dk/product/senhus-connect-panasonic-ac/

For now I could only create CZTACG1 version as it is the version I have at home and can test. If someone is interested in testing DNSKP11 please feel free to reach out.

wrcvr4, 2025-05-15:
Do you ship to Australia by chance ?

senhus, 2025-05-16:
> Do you ship to Australia by chance ?

@wrcvr4 I would love to but it suddenly gets super expensive due to handling fees from the post carriers for customs and etc. I plan to create easy step by step tutorials, maybe getting into electronics will be something for you :-)

marnixt, 2025-05-29:
Can i use the senhus with my CS-Z35XKEW. I have another s32 but it's broken. The airco has a cn cnt port