DomiStyle/esphome-panasonic-ac #144: Deprecation Warnings: `SELECT_SCHEMA`, `SWITCH_SCHEMA`, and `CLIMATE_SCHEMA` – Update Needed Before ESPHome 2025.11.0 (Estimated November 2025)

Boilerplate4u, 2025-06-23:
Hi, I am using the `esphome-panasonic-ac` component with ESPHome 2025.6.0 and I am getting several deprecation warnings in the logs related to schema usage:

```
WARNING Using `select.SELECT_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `select.select_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: panasonic_ac
WARNING Using `switch.SWITCH_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `switch.switch_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: panasonic_ac
WARNING Using `climate.CLIMATE_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `climate.climate_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: panasonic_ac
```

As noted in the warning messages and in the [[ESPHome developer blog post](https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/)]. these deprecated schema functions will be removed in ESPHome 2025.11.0, which is expected to be released in **November 2025**. After that release, this component will stop working unless it is updated to use the new `select.select_schema(...)`, `switch.switch_schema(...)`, and `climate.climate_schema(...)` patterns.

I saw a similar issue posted here recently, but I am not sure if it is about exactly the same problem. If it is, feel free to just mark this post as solved.

Could you please update the component accordingly before ESPHome 2025.11.0 is released? This will help all users avoid potential breakage towards the end of 2025.

Thank you for your great work on this component!

Boilerplate4u, 2025-07-04:
> hey can you please test [#146](https://github.com/DomiStyle/esphome-panasonic-ac/pull/146) and see if it works? thanks

Hey! Sorry for the late reply, I somehow missed GitHub’s notification. I’ll give it a try and get back to you. Cheers.

Boilerplate4u, 2025-07-04:
@cydia2020: I tried your PR on my own test branch, “PR146” (https://github.com/Boilerplate4u/esphome-panasonic-ac/tree/PR146). Works like a charm, thanks a lot!

profigriller, 2025-07-13:
stupid question: What or who is required to create a new release?

closed 2025-07-17 by DomiStyle